realthunder/FreeCAD_assembly3 #1213: error message when "mousing" over constraints in master 0.20

bambukouk, 2022-04-16:
It only happens in master branch (not in link branch).
When "mousing" over/selecting constraints in an assembly, one gets an error message.
[image]
[image]

```
OS: elementary OS 6.1 JAlnir (Pantheon/pantheon)
Word size of FreeCAD: 64-bit
Version: 0.20.28697 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 37ed6fb)
Hash: 37ed6fb92e2d336be0d934530cabc1e5eb815e0a
Python 3.9.12, Qt 5.12.9, Coin 4.0.0, OCC 7.5.3
Locale: English/United Kingdom (en_GB)

```

realthunder, 2022-04-16:
Please update Assembly3 workbench. You can do it using addon manager.

bambukouk, 2022-04-17:
Mine was so old that it didn't even show as an addon in addon manager (I have installed Assembly 3 in master branch manually long time ago...)
but... installing it again via addon manager was easy, painless and solved the problem - thank you!

closed 2022-04-17 by bambukouk