realthunder/FreeCAD_assembly3 #1218: In sketcher / part design text of diameter measurment not shown

hloew, 2022-04-19:
Hello,
first of all this is the information from the "About" (or "Über FreeCAD Link Branch" in German)
OS: Ubuntu 20.04.4 LTS (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.406.28575 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 0ac46d957b0b98af48c7a94dde51e518d1388f31
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

If I draw in the part design workbench a sketch with a circle and set the diameter constraint the text isn't completely invisible but truncated, constraint for radius works fine. Is this an effect of my environment (I don't have the possibility to compare with Windows or another Linux) or a known problem?

[image]

luzpaz, 2022-04-19:
Can you reproduce this on LinkDaily as well ?

realthunder, 2022-04-20:
I can't reproduce the problem. Can you please send me your configurations through `Tools -> Edit parameters -> Export`.

hloew, 2022-04-20:
[export.FCParam.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8524152/export.FCParam.zip)

Hello,
thanks for the replies, I've added the exported parameters and the test with FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220407-glibc2.12-x86_64.AppImage has as result the same error.

With FreeCAD_0.19.3-Linux-Conda_glibc2.12-x86_64.AppImage the diameter text is a little bit truncated at the bottom but readable, with FreeCAD_weekly-builds-28717-Linux-Conda_glibc2.12-x86_64.AppImage the diameter constraint is OK.

My Ubuntu is a Kubuntu and today there was a possibility for a short test withFreeCAD-asm3-Stable-Conda-Py3-Qt5-20220406-glibc2.12-x86_64.AppImage  on a Ubuntu with KDE-Deskktop and on this system the diameter dimension is correct. So I'm afraid this problem comes from the OS and not from the application but on the other hand an AppImage should avoid such things.

Best regards

Hannes