realthunder/FreeCAD_assembly3 #1220: Blank Start Page

QuakeString, 2022-04-23:
There is nothing in start page

OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
[image]

realthunder, 2022-04-25:
Are there any error message in the report view?

QuakeString, 2022-04-25:
No error at all.
[image]

realthunder, 2022-04-25:
Can you please zip the directory `/tmp/FreeCADStartThumbnails` and posted it here?

swebra, 2022-04-26:
I'm also a KDE Manjaro user and am seeing the same thing. I can replicate the issue in both AppImages releases (latest and stable), but the snap release seems to be fine

<details>
<summary>Details of AppImage Installations</summary>

OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

OS: Manjaro Linux (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.407.28576 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 599f7663af69d3dbd77063e834b12f26da513b2d
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
</details>

<details>
<summary>Details of Snap Installation</summary>

OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 2022.424.14555 (Git shallow)
Build type: Release
Branch: (HEAD detached at 2022.04.24-edge)
Hash: 17d71e75693a394d91bf63a918753df49a688426
Python version: 3.8.10
Qt version: 5.15.3
Coin version: 4.0.1
OCC version: 7.6.1
Locale: English/United States (en_US)
</details>

[FreeCADStartThumbnails.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8559431/FreeCADStartThumbnails.zip)

ElPatoNoEsTu, 2022-04-26:
@QuakeString 
Did you explictely check your  "Report - View" Window for any errors ? 

Might it be that this is (too) a "python version" - locale/language handling" related problem?
(  Doing some research on the utf-8 error i stumbled a couple of times over messages stating that there might be connection. )

QuakeString, 2022-04-26:
> Can you please zip the directory `/tmp/FreeCADStartThumbnails` and posted it here?
here it is.
[FreeCADStartThumbnails.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8560188/FreeCADStartThumbnails.zip)

QuakeString, 2022-04-26:
> @QuakeString Did you explictely check your "Report - View" Window for any errors ?
>
> Might it be that this is (too) a "python version" - locale/language handling" related problem? ( Doing some [...]

My report view has no error as i posted the screenshot [here](https://github.com/realthunder/FreeCAD_assembly3/issues/1220#issuecomment-1108863897)
and here is my terminal output, absolutely no error.
[image]

realthunder, 2022-04-26:
It may be related to [this](https://bugs.launchpad.net/ubuntu/+source/glibc/+bug/1939993) issue. If so, we may need to add a patch to conda qt recipe. @looooo can you please check this out? I see upstream also [reporting](https://forum.freecadweb.org/viewtopic.php?f=4&t=66363&start=60#p579284) similar problem.