realthunder/FreeCAD_assembly3 #1221: 'utf-8' codec can't decode byte 0xfc in position 301: invalid start byte

ElPatoNoEsTu, 2022-04-23:
Hi all, 

after replacing a former version (mid summer 2021)  by the current 2022.04.07  i get the above message. 

doinge a search - i found it might have to do with th "locale" setting -  but I cant  "export"  encodiung-setting due to using windows as os .   

Also it looks like the startup-scrip isn't read correctely as i get a blank "freecad-window" without the used possibilitiys in selecting a last used file ... 

any Idea ? 

Regards, Stefan

PS Unfortunately I even can't tell more about the file - where this happens.   ( are there any "debug" switches to tell freecad to be more verbose regarding its startup ? )

luzpaz, 2022-04-23:
Where do you get the message? Do you have an example file?
If you can load FreeCAD, please post your full [About](https://wiki.freecad.org) info

ElPatoNoEsTu, 2022-04-23:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.407.28576 (Git)
Build type: Release
Branch: LinkStage3
Hash: 599f7663af69d3dbd77063e834b12f26da513b2d
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: German/Germany (de_DE)
[image]

luzpaz, 2022-04-23:
Do you see this in LinkDaily as well?

realthunder, 2022-04-24:
Can you please send me your user configuration file, `Tools -> Edit parameters -> Export`.

ElPatoNoEsTu, 2022-04-24:
[user_config_file.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8550449/user_config_file.zip)

realthunder, 2022-04-25:
Can you please zip the following directory and post it here? Also please try [daily branch](https://github.com/realthunder/FreeCAD_assembly3/releases/download/0.11/FreeCAD-asm3-Daily-Win64-Py3-Qt5-20220407.7z) and see if works.

```
C:\Users\<your_username>\AppData\Local\Temp\FreeCADStartThumbnails
```

ElPatoNoEsTu, 2022-04-25:
By the way - I also noticed a blank Start-Screen. 

[FreeCADStartThumbnails.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8555786/FreeCADStartThumbnails.zip)

closed 2022-04-25 by realthunder