realthunder/FreeCAD_assembly3 #1225: Assembly3 can not be loaded

Nemo-HelloWorld, 2022-04-29:
### When I run FreeCAD, it reports the following error:

```
During initialization the error "

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.8 from "D:\FreeCAD\Build\bin\FreeCAD_d.exe"
  * The NumPy version is: "1.19.3"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: No module named 'numpy.core._multiarray_umath'
" occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "D:\FreeCAD\Build\bin\lib\site-packages\numpy\core\__init__.py", line 22, in <module>
    from . import multiarray
  File "D:\FreeCAD\Build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "D:\FreeCAD\Build\bin\lib\site-packages\numpy\core\multiarray.py", line 12, in <module>
    from . import overrides
  File "D:\FreeCAD\Build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "D:\FreeCAD\Build\bin\lib\site-packages\numpy\core\overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
  File "D:\FreeCAD\Build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
ModuleNotFoundError: No module named 'numpy.core._multiarray_umath'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 156, in InitApplications
  File "D:\FreeCAD\Build\bin\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\29173\AppData\Roaming\FreeCAD\Mod\Assembly3\freecad\asm3\init_gui.py", line 3, in <module>
    from . import gui
  File "D:\FreeCAD\Build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\29173\AppData\Roaming\FreeCAD\Mod\Assembly3\freecad\asm3\gui.py", line 10, in <module>
    from .utils import getElementPos,objName,addIconToFCAD,guilogger as logger
  File "D:\FreeCAD\Build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\29173\AppData\Roaming\FreeCAD\Mod\Assembly3\freecad\asm3\utils.py", line 11, in <module>
    import numpy as np
  File "D:\FreeCAD\Build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "D:\FreeCAD\Build\bin\lib\site-packages\numpy\__init__.py", line 140, in <module>
    from . import core
  File "D:\FreeCAD\Build\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "D:\FreeCAD\Build\bin\lib\site-packages\numpy\core\__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.8 from "D:\FreeCAD\Build\bin\FreeCAD_d.exe"
  * The NumPy version is: "1.19.3"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: No module named 'numpy.core._multiarray_umath'
```
### System information

- **FreeCAD version: 0.19

- **Operating system: Windows 10

luzpaz, 2022-04-29:
> **FreeCAD version: 0.19

Please always add your full [About](https://wiiki.freecad.org/About) info next time. You should also test this in [v0.20dev](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

Nemo-HelloWorld, 2022-04-30:
> > **FreeCAD version: 0.19
> 
> Please always add your full [About](https://wiiki.freecad.org/About) info next time. You should also test this in [v0.20dev](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

Done :)

luzpaz, 2022-06-30:
@Nemo-HelloWorld please test with v0.20 stable

Nemo-HelloWorld, 2022-07-01:
> @Nemo-HelloWorld please test with v0.20 stable

Works fine in v0.20.

closed 2022-07-01 by luzpaz