realthunder/FreeCAD_assembly3 #1227: Addon manager (workbenches, macros) not listing

andyylin, 2022-05-04:
Hi, I'm unable to access any addons through the add-on manager. I'm able to access add-ons on the 0.20 standard build, but not on realthunder's daily build.

```
OS: macOS 12.3
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git)
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.9.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: C/Default (C)
```

Here are the Report View logs:
```
15:14:21  During initialization the error "[Errno 2] No such file or directory: '"/Applications/FreeCAD realthunder.app/Contents/MacOS/pythonLink" -m pip install "/Users/andylin/Library/Preferences/FreeCAD/Mod/InventorLoader/libs/olefile-0.46.zip"'" occurred in /Users/andylin/Library/Preferences/FreeCAD/Mod/InventorLoader/InitGui.py
15:14:21  Please look into the log file for further information
15:14:25  1.444711 <asm3.main> init_gui.py(14): no solver backend found
15:17:31  Traceback (most recent call last):
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1346, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1285, in request
    self._send_request(method, url, body, headers, encode_chunked)
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1331, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1280, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1040, in _send_output
    self.send(msg)
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 980, in send
    self.connect()
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1454, in connect
    self.sock = self._context.wrap_socket(self.sock,
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/ssl.py", line 1040, in _create
    self.do_handshake()
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
15:17:31  ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)
15:17:31  
During handling of the above exception, another exception occurred:

15:17:31  Traceback (most recent call last):
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/Mod/AddonManager/addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1349, in do_open
    raise URLError(err)
15:17:31  urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)>
15:17:31  Failed to get url https://raw.githubusercontent.com/FreeCAD/FreeCAD-addons/master/addonflags.json
15:17:31  Debug: addon_flags.json not found
15:17:31  Traceback (most recent call last):
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1346, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1285, in request
    self._send_request(method, url, body, headers, encode_chunked)
15:17:31    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/http/client.py", line 1331, in _send_request
[... 56 lines cut ...]
15:17:35    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
15:17:35  ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)
15:17:35  
During handling of the above exception, another exception occurred:

15:17:35  Traceback (most recent call last):
15:17:35    File "/Applications/FreeCAD realthunder.app/Contents/Mod/AddonManager/addonmanager_utilities.py", line 126, in urlopen
    u = urllib2.urlopen(req, timeout=timeout)
15:17:35    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
15:17:35    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
15:17:35    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
15:17:35    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
15:17:35    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
15:17:35    File "/Applications/FreeCAD realthunder.app/Contents/lib/python3.9/urllib/request.py", line 1349, in do_open
    raise URLError(err)
15:17:35  urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)>
15:17:35  Failed to get url https://wiki.freecadweb.org/Macros_recipes
15:17:35  Appears to be an issue connecting to the Wiki, therefore cannot retrieve Wiki macro list at this time
```

realthunder, 2022-05-04:
Can you please run the following code in FreeCAD Python console, and post here the output?

```python
import os
os.environ['SSL_CERT_FILE']
```

andyylin, 2022-05-05:
> Can you please run the following code in FreeCAD Python console, and post here the output?

Yes, thank you, here's the output.

```
>>> import os
>>> os.environ['SSL_CERT_FILE']
'/Applications/FreeCAD realthunder.app/Contents/lib/python3*/site-packages/certifi/cacert.pem'
>>> 
```

cadmarx, 2022-05-14:
I am having the same issue on the below release. Any help is appreciated.

OS: macOS 12.3
Word size of FreeCAD: 64-bit
Version: 2022.407.26244 +4693 (Git)
Build type: Release
Branch: LinkDaily
Hash: 9610830a8fd492b3b3d96f8328507c80f1437cda
Python version: 3.9.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: C/Default (C)

andyylin, 2022-06-13:
I don't know why, but for some reason a full clean install allowed me to see the add-on manager, but now it's blank again. Unfortunately, it's been a while so hard to know what might have changed in between.

luzpaz, 2022-07-08:
Specifically
```
09:20:09  urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)>
09:20:09  Failed to get url https://wiki.freecadweb.org/Macros_recipes
09:20:09  Appears to be an issue connecting to the Wiki, therefore cannot retrieve Wiki macro list at this time
```