realthunder/FreeCAD_assembly3 #1228: Base planes used as Elements in Assembly 3 WB

jpg87Fr, 2022-05-05:
Hello,
it seems that the base planes used as Elements in Assembly 3 WB all 3 behave like XY in the "official" branch of FreeCAD, while this problem has been fixed for a while in Link Branch. Where can this come from ?

Nemesis81, 2022-10-04:
@realthunder I can confirm the issue recently
https://forum.freecadweb.org/viewtopic.php?f=20&t=72256&p=630621#p630621

Marodut, 2023-04-26:
@realthunder I have encountered the same issue recently:

https://forum.freecad.org/viewtopic.php?p=677585#p677585

```
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32985 (Git)
Build type: Release
Branch: master
Hash: 5cd377f931312ede305e4c764ba2028b20eaacfd
Python 3.8.16, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.11.4
  * Behave-Dark-Colors 0.1.1
  * CurvedShapes 1.0.4
  * Curves 0.6.8
  * Dracula 0.0.4
  * ExplodedAssembly
  * fasteners 0.4.56
  * fcgear 1.0.0
  * freecad.gears 1.0.0
  * OpticsWorkbench 1.0.9
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.63
```