realthunder/FreeCAD_assembly3 #1231: Issues with daily ver 2022.04.30

ChellSol, 2022-05-10:
Hi

First let me say freecad is brilliant.
This is the first time posting so my apologies if I have posed in the wrong area.

I’ve been playing with daily ver 2022.04.30 and had the following issues.

•	Text in the tree view comb box does not show any text on open when text size is set to anything other than 0 In preferences, general, tree view, font size. Changing the text size fixes the combo box, until freecad is restarted.
•	In sketcher when you select and right clicking on multiple constraints within the combo box and select hide it doesn’t refresh the combo box. Each constraint still shows the visibility ticks next to them.
•	Using my laptop with the mouse in touchpad mode. When I hold down shift & move mouse pad. Sometimes it moves, sometimes it pans and sometimes it does nothing. This occurs at the same place within freecad. So, if you keep trying the same mouse movement you will eventually get the one you want.
•	The new right click popup menu transparency can make the menu unreadable in some cases.
•	I have had a memory leak (I think) occur when creating multitransform pattern. Cpu was running a 50% and the ram went to 100% 16gb.  This kept going until the program crashed. The pattern consisted of 2 linear patterns. Just a simple grid of holes. Problem is it appears to be random.

Feature request.
•	On my windows laptop I have a 4K screen, it defaults to text scaling of 300%. This makes multiple screens in freecad difficult to use. Ideally freecad would have setting that would allow the freecad windows/text/panels etc to be resized without needing to turn off the windows text scaling.
•	The ability to temporarily disable an item such as a multitransform pattern from recalculating automatically. While having the rest of the file calculating as normal.

luzpaz, 2022-05-10:
Welcome. Thanks for reporting. 

First, please always post your full [About](https://wiki.freecad.org/About) info

Limit one bug or feature per ticket. More than that will quickly get lost.

Try to be as descriptive and illustrative as possible. Add screenshots of what you're referring to. When reporting issues, please try to think of the people that are reading the reports. Their time is important and so if you can make it really easy to comprehend, the higher the probability that you may get your needs met.

bambukouk, 2022-05-16:
This is not an assembly issue?