realthunder/FreeCAD_assembly3 #1233: Image workbench: SEGFAULT when simply doing "Create image plane..."

ardugitter, 2022-05-14:
The title explains it. To reproduce, switch to Image workbench, click on the "Create image plane..." icon and load this image:
[image]

A SEGFAULT will happen every time. Here is the console output:

```
FreeCAD 0.20, Libs: 2022.430R26244 +4758 (Git)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bd60) [0x7f3912abbd60]
#1  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x14c7fa) [0x7f39083fd7fa]
#2  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x1499ff) [0x7f39083fa9ff]
#3  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x1e686e) [0x7f390849786e]
#4  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x1e6ffb) [0x7f3908497ffb]
#5  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x2206fc) [0x7f39084d16fc]
#6  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x2155ee) [0x7f39084c65ee]
#7  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x47b8de) [0x7f390872c8de]
#8  0x7f391529139d in NaviCubeShared::drawNaviCube(SoCamera*, bool, int, bool) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libFreeCADGui.so+0x13cd
#9  0x7f39152917b6 in NaviCubeImplementation::drawNaviCube(bool) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libFreeCADGui.so+0x46
#10  0x7f391525ba45 in Gui::View3DInventorViewer::renderScene() from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libFreeCADGui.so+0x515
#11  0x7f391520ac0d in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libFreeCADGui.so+0x31d
#12  0x7f3915210ec5 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libFreeCADGui.so+0x45
#13  0x7f3913a8595d in QWidget::event(QEvent*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x5a3
#14  0x7f3913aeedb7 in QFrame::event(QEvent*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x27
#15  0x7f3913c69ebe in QGraphicsView::viewportEvent(QEvent*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x456
#16  0x7f3912ff0019 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Core.so.5+0xcd
#17  0x7f3913a5a7e1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0xcb
#18  0x7f3913a6031d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#19  0x7f3914dbd5f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libFreeCADGui.so+0x89
#20  0x7f3912ff2606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Core.so.5+0x86
#21  0x7f3913a80612 in QWidgetPrivate::sendPaintEvent(QRegion const&) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x28
#22  0x7f3913a80d3f in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x707
#23  0x7f3913a814b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#24  0x7f3913a811be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#25  0x7f3913a814b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#26  0x7f3913a811be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#27  0x7f3913a814b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#28  0x7f3913a811be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#29  0x7f3913a814b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#30  0x7f3913a811be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#31  0x7f3913a814b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#32  0x7f3913a811be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#33  0x7f3913a814b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#34  0x7f3913a811be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#35  0x7f3913a814b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#36  0x7f3913a813ce in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x1de
#37  0x7f3913a813ce in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x1de
#38  0x7f3913a813ce in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x1de
#39  0x7f3913a813ce in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x1de
#40  0x7f3913a813ce in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x1de
#41  0x7f3913a813ce in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x1de
#42  0x7f3913a813ce in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x1de
#43  0x7f3913a813ce in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0x1de
#44  0x7f3913a811be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#45  /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5(+0x17ed08) [0x7f3913a64d08]
#46  /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5(+0x17ef01) [0x7f3913a64f01]
#47  /tmp/.mount_FreeCAMLfnpM/usr/bin/../lib/libQt5Widgets.so.5(+0x1af9e2) [0x7f3913a959e2]
[... cut ...]
#64  /tmp/.mount_FreeCAMLfnpM/usr/bin/FreeCADLink(+0x470d) [0x563b278d270d]
#65  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xea) [0x7f3912aa6d0a]
#66  /tmp/.mount_FreeCAMLfnpM/usr/bin/FreeCADLink(+0x4aa9) [0x563b278d2aa9]
```

Here is my system info:

```
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United Kingdom (en_GB)
```

Thank you!

realthunder, 2022-05-14:
Can you please try [this](https://drive.google.com/file/d/16ZVb6toX4xG09SqQbWbjCM-3Pozo-xwv/view?usp=sharing) version?

ardugitter, 2022-05-14:
Unfortunately that crashes in the same way too.
```

$ FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220511-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCAipF3cg
FreeCAD 0.20, Libs: 2022.511R26244 +4790 (Git)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bd60) [0x7f0646239d60]
#1  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x14c7fa) [0x7f06374097fa]
#2  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x1499ff) [0x7f06374069ff]
#3  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x1e686e) [0x7f06374a386e]
#4  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x1e6ffb) [0x7f06374a3ffb]
#5  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x2206fc) [0x7f06374dd6fc]
#6  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x2155ee) [0x7f06374d25ee]
#7  /usr/lib/x86_64-linux-gnu/dri/i965_dri.so(+0x47b8de) [0x7f06377388de]
#8  0x7f0648a12515 in NaviCubeShared::drawNaviCube(SoCamera*, bool, int, bool) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libFreeCADGui.so+0x15f5
#9  0x7f0648a12816 in NaviCubeImplementation::drawNaviCube(bool) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libFreeCADGui.so+0x46
#10  0x7f06489dc9b5 in Gui::View3DInventorViewer::renderScene() from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libFreeCADGui.so+0x515
#11  0x7f064898bb7d in SIM::Coin3D::Quarter::QuarterWidget::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libFreeCADGui.so+0x31d
#12  0x7f0648991e35 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::paintEvent(QPaintEvent*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libFreeCADGui.so+0x45
#13  0x7f064720395d in QWidget::event(QEvent*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x5a3
#14  0x7f064726cdb7 in QFrame::event(QEvent*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x27
#15  0x7f06473e7ebe in QGraphicsView::viewportEvent(QEvent*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x456
#16  0x7f064676e019 in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Core.so.5+0xcd
#17  0x7f06471d87e1 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0xcb
#18  0x7f06471de31d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#19  0x7f064853c3c9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libFreeCADGui.so+0x89
#20  0x7f0646770606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Core.so.5+0x86
#21  0x7f06471fe612 in QWidgetPrivate::sendPaintEvent(QRegion const&) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x28
#22  0x7f06471fed3f in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x707
#23  0x7f06471ff4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#24  0x7f06471ff1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#25  0x7f06471ff4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#26  0x7f06471ff1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#27  0x7f06471ff4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#28  0x7f06471ff1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#29  0x7f06471ff4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#30  0x7f06471ff1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#31  0x7f06471ff4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
#32  0x7f06471ff1be in QWidgetPrivate::drawWidget(QPaintDevice*, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0xb86
#33  0x7f06471ff4b1 in QWidgetPrivate::paintSiblingsRecursive(QPaintDevice*, QList<QObject*> const&, int, QRegion const&, QPoint const&, int, QPainter*, QWidgetBackingStore*) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Widgets.so.5+0x2c1
[... cut ...]
#62  0x7f064676d019 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Core.so.5+0x187
#63  0x7f0646770e8f in QCoreApplication::exec() from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libQt5Core.so.5+0xfd
#64  0x7f06484ab333 in Gui::Application::runApplication() from /tmp/.mount_FreeCAipF3cg/usr/bin/../lib/libFreeCADGui.so+0x443
#65  /tmp/.mount_FreeCAipF3cg/usr/bin/FreeCADLink(+0x470d) [0x55754b2b470d]
#66  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xea) [0x7f0646224d0a]
#67  /tmp/.mount_FreeCAipF3cg/usr/bin/FreeCADLink(+0x4aa9) [0x55754b2b4aa9]
```

realthunder, 2022-05-15:
Can you please post here your user configurations, `Tools -> Edit parameters -> Export`.

ardugitter, 2022-05-15:
Sure, here it is:

[params.log](https://github.com/realthunder/FreeCAD_assembly3/files/8695726/params.log)

I gave it a .log extension so it would upload. Thanks

realthunder, 2022-05-16:
Still can't reproduce. Is it possible for you to install dbg symobl of `i965_dri.so`, and post the crash stack again?

realthunder, 2022-05-16:
Don't know your Linux distro. Try search something like `mesa debug symbols`

ardugitter, 2022-05-27:
I will close the issue as I do not have access to the debug symbols.

I'd just like to add that I do not see this crash in the main branch of FreeCAD 0.19.3 so maybe we can just keep this problem in mind in case someone else sees it in the future.

Thanks.

closed 2022-05-27 by ardugitter

luzpaz, 2022-05-28:
@ardugitter have you tried against [v0.20dev](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) ?

ardugitter, 2022-05-28:
> @ardugitter have you tried against [v0.20dev](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds) ?

I just tried that and the crash does not occur with this either.

I have to use the RT branch though  to avoid TNP and because I love Assembly3.