realthunder/FreeCAD_assembly3 #1243: [fixed] Appimage refuses to run, libGL error: MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus dri.so: appimage Debian

snelweg, 2022-06-05:
Works in 'official'  AppImages, only the LinkStage3 appimages do  not work on my machine. I am using Debian SID because I need to use LinuxCNC on this machine.

`$ Downloads/FreeCAD-asm3-Daily-Conda-Py3-Qt5-20220430-glibc2.12-x86_64.AppImage 
/tmp/.mount_FreeCA3068yI
FreeCAD 0.20, Libs: 2022.430R26244 +4758 (Git)
Sheet Metal workbench loaded
Assembly4 workbench (v0.11.12) loaded
libGL error: MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: kan gedeeld objectbestand niet openen: Bestand of map bestaat niet (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: kan gedeeld objectbestand niet openen: Bestand of map bestaat niet (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: kan gedeeld objectbestand niet openen: Bestand of map bestaat niet (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
Initializing Assembly4 workbench (v0.11.12) ............................ done.
<PropShape> PropertyTopoShape.cpp(414): Pending recompute for generating element map: imageimport#Body
<PropShape> PropertyTopoShape.cpp(414): Pending recompute for generating element map: imageimport#Sketch
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x38920) [0x7f4e26b75920]
#1  0x7f4e276c08a6 in QOpenGLContext::isValid() const from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libQt5Gui.so.5+0
#2  0x7f4e292d3065 in SIM::Coin3D::Quarter::QuarterWidgetP::removeFromCacheContext(QuarterWidgetP_cachecontext*, QOpenGLWidget const*) from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libFreeCADGui.so+0xc5
#3  0x7f4e292d313a in SIM::Coin3D::Quarter::QuarterWidgetP::~QuarterWidgetP() from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libFreeCADGui.so+0x1a
#4  0x7f4e292cdac9 in SIM::Coin3D::Quarter::QuarterWidget::~QuarterWidget() from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libFreeCADGui.so+0xb9
#5  0x7f4e29320e0a in Gui::View3DInventorViewer::~View3DInventorViewer() from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libFreeCADGui.so+0xa
#6  0x7f4e29300c7d in Gui::View3DInventor::~View3DInventor() from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libFreeCADGui.so+0x17d
#7  0x7f4e29300eba in Gui::View3DInventor::~View3DInventor() from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libFreeCADGui.so+0xa
#8  0x7f4e270ce54e in QObject::event(QEvent*) from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libQt5Core.so.5+0x76
#9  0x7f4e27b4d07f in QWidget::event(QEvent*) from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libQt5Widgets.so.5+0xcc5
#10  0x7f4e27bf734f in QMainWindow::event(QEvent*) from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libQt5Widgets.so.5+0xdd
#11  0x7f4e27b217f2 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libQt5Widgets.so.5+0xdc
#12  0x7f4e27b2731d in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libQt5Widgets.so.5+0x1ac9
#13  0x7f4e28e845f9 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libFreeCADGui.so+0x89
#14  0x7f4e270b9606 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libQt5Core.so.5+0x86
#15  0x7f4e270b983e in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libQt5Core.so.5+0x1da
#16  0x7f4e270b9eb2 in QCoreApplication::exec() from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libQt5Core.so.5+0x120
#17  0x7f4e28df3563 in Gui::Application::runApplication() from /tmp/.mount_FreeCA3068yI/usr/bin/../lib/libFreeCADGui.so+0x443
#18  /tmp/.mount_FreeCA3068yI/usr/bin/FreeCADLink(+0x470d) [0x55e1db08870d]
#19  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xcd) [0x7f4e26b607fd]
#20  /tmp/.mount_FreeCA3068yI/usr/bin/FreeCADLink(+0x4aa9) [0x55e1db088aa9]
`

Fix is here: 

`https://forum.manjaro.org/t/many-appimages-have-stopped-working-after-todays-update/69445/20`

> OS: Debian GNU/Linux bookworm/sid (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2022.430.26244 +4758 (Git)
Build type: Release
Branch: LinkDaily
Hash: b024b876480a533018fdc702145e9e38f8c54ba6
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)

```
$ uname -a
Linux DESKTOP-9FIBCI1 5.17.0-1-rt-amd64 #1 SMP PREEMPT_RT Debian 5.17.3-1 (2022-04-18) x86_64 GNU/Linux
```

```
$ cat /etc/debian_version 
bookworm/sid
```

luzpaz, 2022-07-04:
@realthunder this is the known issue with AppImage bundling deprecated `libstdc++.so.6`  
Hey @looooo what's the workaround we use on upstream appimage for this?

looooo, 2022-07-04:
I guess conda-forge updated the libraries which allow to use newer glibc version. I did not do anything in this regard :). There is also the other workaround, which I would like to add one day. (Can't find issue right now)

luzpaz, 2022-07-04:
@Realthunder are you using the latest condaforge libs ?

mwlandolsi, 2022-08-03:
After changing from mint 20 to mint 21 The older versions of FreeCAD linkstage could not be launched. The last version, Appimage 09-07-2022 can start, but the start page is blank. You know, there are many works made with older versions that may cause errors in newer version. So I keep older versions of Freecad to be able to open, eventually, my old files.
OS: Linux Mint 21 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
[image]

Derek19168284, 2023-01-24:
Hello Everyone, I have a similar trouble with the Linux Mint 21.1 Cinnamon. 
I'm studing python and I programming on visual studio code operated from Anaconda. 
It works almost completely, but when I launch a program with graphics, on the terminal appear these errors:

libGL error: MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: impossibile aprire il file oggetto condiviso: File o directory non esistente (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open crocus: /usr/lib/dri/crocus_dri.so: impossibile aprire il file oggetto condiviso: File o directory non esistente (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: impossibile aprire il file oggetto condiviso: File o directory non esistente (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast

can someone suggest to me how to arreglate this issue? Is related to my Radeon Video card?
Thank you very much for your help!!!
DEREK

daniil-loban, 2023-04-15:
@Derek19168284  did you solve it?

kaktusus, 2023-08-06:
After updating my system no AppImage old or new starts up.
I use AppImage weekly-builds.

```
$ cat /etc/debian_version 
trixie/sid
```

```
$ uname -a
Linux programator-2 6.4.0-1-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.4.4-2 (2023-07-30) x86_64 GNU/Linux
```

After doing a few steps I get errors in the console:
```
$ ./FreeCAD_weekly-builds-33719-2023-08-06-conda-Linux-x86_64-py310.AppImage
FreeCAD 0.22.0, Libs: 0.22.0devR33719 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
Sheet Metal workbench loaded
libGL error: MESA-LOADER: failed to open crocus: /tmp/.mount_FreeCAhgE0x2/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.32 not found (required by /lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open crocus: /tmp/.mount_FreeCAhgE0x2/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.32 not found (required by /lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: crocus
libGL error: MESA-LOADER: failed to open swrast: /tmp/.mount_FreeCAhgE0x2/usr/bin/../lib/libstdc++.so.6: version GLIBCXX_3.4.32 not found (required by /lib/x86_64-linux-gnu/libLLVM-15.so.1) (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
WebEngineContext used before QtWebEngine::initialize() or OpenGL context creation failed.
QGLXContext: Failed to create dummy context
Failed to create OpenGL context for format QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize 24, redBufferSize -1, greenBufferSize -1, blueBufferSize -1, alphaBufferSize -1, stencilBufferSize 8, samples 0, swapBehavior QSurfaceFormat::DefaultSwapBehavior, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile) 
/tmp/.mount_FreeCAhgE0x2/AppRun: linia 43:  6336 Przerwane               (zrzut pamięci) ${MAIN} "$@"
```

Use the following to check support for GLIBCXX_3.4.32 and renew linking.

```
$ strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX
$ ls -la /usr/lib/x86_64-linux-gnu/libstdc++.so.6

```

Before I did the following linking in my system, the error messages looked completely different.

``` 
ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0032 /usr/lib/dri/
ln -s /usr/lib/x86_64-linux-gnu/dri /usr/lib/
```

kaktusus, 2023-08-10:
**FreeCAD_weekly-builds-33774-2023-08-10-conda-Linux-x86_64-py310.AppImage**

there are still problems running AppImage
If the solution is known and it relies on replacing one library file _(libstdc++.so.6)_ why has it not been implemented to life?
:sob:

kaktusus, 2023-08-14:
**FreeCAD_weekly-builds-33798-2023-08-13-conda-Linux-x86_64-py310.AppImage**

there are still problems running AppImage

kaktusus, 2023-08-21:
A second convenient workaround to a problem that still exists:
https://forum.freecad.org/viewtopic.php?style=4&p=700883#p700883