realthunder/FreeCAD_assembly3 #1246: Render WB incompatibilities

goeland86, 2022-06-15:
I've been having a bit of a hard time getting my assembly rendered from within FreeCAD, using the Render WB.
My project (https://github.com/goeland86/MonsterKossel/) uses the Assembly3 and Fasteners WB to create the full project.

When I tried recently to render the final assembly in FreeCAD using the Render WB I found an issue that I traced to the Fasteners WB. This has since been fixed. (See forum post here: https://forum.freecadweb.org/viewtopic.php?f=8&t=69321&p=602554#p602554)

I'd like to report that at present I cannot render my project by simply creating a view on the master assembly. If I do, I get the following error: 
```
21:34:29  Running the Python command 'Render_Render' failed:
Traceback (most recent call last):
  File "C:\Users\goela\AppData\Roaming\FreeCAD\Mod\Render\Render\commands.py", line 185, in Activated
    project.Proxy.render()
  File "C:\Users\goela\AppData\Roaming\FreeCAD\Mod\Render\Render\project.py", line 441, in render
    objstrings = [
  File "C:\Users\goela\AppData\Roaming\FreeCAD\Mod\Render\Render\project.py", line 442, in <listcomp>
    get_rdr_string(v)
  File "C:\Users\goela\AppData\Roaming\FreeCAD\Mod\Render\Render\rdrhandler.py", line 269, in get_rendering_string
    return RendererHandler._render_object(self, name, view)
  File "C:\Users\goela\AppData\Roaming\FreeCAD\Mod\Render\Render\rdrhandler.py", line 352, in _render_object
    rends = renderables.get_renderables(
  File "C:\Users\goela\AppData\Roaming\FreeCAD\Mod\Render\Render\renderables.py", line 103, in get_renderables
    renderables = _get_rends_from_plainapplink(
  File "C:\Users\goela\AppData\Roaming\FreeCAD\Mod\Render\Render\renderables.py", line 249, in _get_rends_from_plainapplink
    base_rends = get_renderables(linkedobj, name, material, mesher, **kwargs)
  File "C:\Users\goela\AppData\Roaming\FreeCAD\Mod\Render\Render\renderables.py", line 142, in get_renderables
    renderables = _get_rends_from_partfeature(
  File "C:\Users\goela\AppData\Roaming\FreeCAD\Mod\Render\Render\renderables.py", line 473, in _get_rends_from_partfeature
    color = _get_shapecolor(obj, transparency_boost)
  File "C:\Users\goela\AppData\Roaming\FreeCAD\Mod\Render\Render\renderables.py", line 505, in _get_shapecolor
    vobj.ShapeColor[0],

'Gui.ViewProviderLink' object has no attribute 'ShapeColor'
```

I'm running this version of Linkstage3:

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.430.28641 (Git)
Build type: Release
Branch: LinkStage3
Hash: a77045488092c24ece25d63c91f362cbd2181d36
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US)

If you download the zip of the repository (or do a git clone --depth=1) you can try and open https://github.com/goeland86/MonsterKossel/blob/master/Assembly-for-render.FCStd (careful, lots of dependencies on almost all files in the root of the repo, and a couple of the STEP files as well, using sub-assemblies in a hierarchy).

I can get further, and potentially even get a render, IF I go through the render project objects and remove any reference to elements or constraints, and the various group projects. It's very tedious. And when I'm done, the render doesn't look right. Here's what I expect (generated with Blender from the STEP import):
[image]

Here's what I obtain when trying to use the Render WB after cleaning out the objects causing the no attribute error above:
[image]

Any input on this is welcome. I have a workaround for now, but as much as I love Blender, I'd like to manage everything from within FreeCAD, since it in theory is able to.

goeland86, 2022-06-15:
The repo contains the render file with all the Assembly3 elements.
The file attached below produces the screenshot above, attached for reference (the dependencies on the files in the repo stays the same).
[Assembly-for-render.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8912751/Assembly-for-render.zip)

luzpaz, 2022-06-30:
CC @howetuft just a heads up that this PR references Render WB