realthunder/FreeCAD_assembly3 #1247: FreeCad 0.20 feature: Sketcher copy / paste (via xml) is not working in this fork

derMart, 2022-06-18:
Hey, it seems the very useful feature of generic copy paste sketcher geometries in between different sketches, even in between different instances of freecad (see here: https://www.youtube.com/watch?v=ZH432giUasM) is not working in this fork.
If I select a portion of a sketch inside sketcher, the Edit -> Copy menu is greyed out.

I am using FreeCad Link Branch daily snap package (Version 2022.617)

derMart, 2022-06-18:
oh dear just realized this is not even merged in 'normal' FreeCad. This is very useful, so I would like to add this feature in this fork as feature request please!

luzpaz, 2022-06-18:
You're referring to the open PR in upstream, right? https://github.com/FreeCAD/FreeCAD/pull/5480