realthunder/FreeCAD_assembly3 #1252: [Feature Request] Add a "Package" functionality to make it easier to share files with others

ToniTen, 2022-07-14:
My current workflow for assemblies is:

1. Create the components, using Part Design or via importing step files, each in a separate file
2. Wrap each component in an Assembly3 container, and declare the "interface" via creating the elements needed to connect the part to other parts
3. Freeze each of these wrapper assemblies
4. Create first level sub assemblies where 2 or more parts are imported via links, dropped into an Assembly3 container, and constrained with Assembly3 constraints, using the elements declared in the wrapper assemblies. Export the elements from the wrapper sub assemblies into the outer sub assembly, if they are needed. Once these sub assemblies are finished, they are frozen.
5. Create higher level sub assemblies composed of first level sub assemblies as necessary, until all components are assembled.

This allows me to create a logical structure, and to maximize reusability, for example if say a given sub assembly is needed 3 times, or if a given sub assembly is shared between multiple variants. This also creates a lot of files for a given final assembly.

If I need to share the FreeCAD files with someone else, I thought that I could just go to Link Actions/Import all links, but this operation fails if any of of the files is partially loaded (due to frozen assemblies). Thus, I'm required to go to all the individual sub assemblies, unfreeze them, and only then I can import all links. I then have to save this file under a new name, and make sure to not save any of the now unfrozen sub-assemblies.

The feature I want is to have an option under Assembly3 to "Package" the assembly by fully loading all components and sub assemblies, importing all links, and then saving that to a new file, then unload all of the now fully loaded files so they can remain partially loaded.

realthunder, 2022-07-15:
You don't need to unfreeze the assembly. Just right click the top document item in the tree view, and select 'Reload document'. This will cause all linked document to be reloaded in full.

ToniTen, 2022-07-15:
> You don't need to unfreeze the assembly. Just right click the top document item in the tree view, and select 'Reload document'. This will cause all linked document to be reloaded in full.

Thanks! that makes things easier, but as I usually work with at least 5 levels of nested assemblies (and have in the past used over 20), I need to reload the document multiple times. Could it be possible to add a "reload document hierarchy" that recursively reloads everything?

ToniTen, 2022-08-04:
I saw that you added the "Reload all documents" functionality to Link branch ([commit](https://github.com/realthunder/FreeCAD/commit/4df02e3a1886418d25be34ed5eae3a9597296248)) . Thanks a lot. That's exactly what I need. Could you send send a pull request so this eventually lands upstream?