realthunder/FreeCAD_assembly3 #1258: Dimensions stored in spreadsheet do not update unles you open the dialog and press cancel

grtyvr, 2022-07-30:
If you have dimensions in a spreadsheet, and update the dimensions in the spreadsheet the dimension will not update in the drawing unless you select the dimension for editing in the document, open the formula edition, press ok, and then press cancel on the dimension dialog.

luzpaz, 2022-07-30:
Always post full [about](https://wiki.freecad.org/About) info please!

grtyvr, 2022-07-31:
My bad.
-----------------8><---------------------------------------
OS: Ubuntu 20.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.111.28429 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: ec814c42d02f15171f9dbb7f650400c9943abfe0
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/Canada (en_CA)

luzpaz, 2022-07-31:
@oficinerobotica do you mind confirming the bug report ?

grtyvr, 2022-07-31:
I have been doing more testing.  It does not always happen.  I am trying to find a set of reproducible steps.

realthunder, 2022-08-01:
@grtyvr Please test using the latest 0709 release.

grtyvr, 2022-08-01:
0709 release.  Steps to reproduce.

Create new document: doc_1.
Create new body.
Create sketch.
Add spreadsheet
Enter dim. in spreadsheet and give it an alias: alias_1

Open new document: doc_2.
Create new body.
Create new sketch
Assign dimension from spreadsheet in other document using <<doc_1>>#Spreadsheet.alias_1

If you change the value in the spreadsheet you will have to open the formula editor in doc_2 that references that alias and click cancel before the new value is used in the drawing.

grtyvr, 2022-08-04:
Might be related but in the pad dialog if you reference a dimension in another document then you have to delete the formula and re-enter it for the new dimension to be used.