realthunder/FreeCAD_assembly3 #1259: GPU hang while editing a sketch

dbozec, 2022-08-10:
I happened to me several time in the last few days.
While editing a sketch I suddenly get a black screen, and the only way to get something back is to reboot my computer.
Here are details from FreeCad about form :
```
OS: Ubuntu 20.04.4 LTS (openbox)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.704.28821 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: c83a0368027dcdb972bfb3e0a00fbf9c26536fcd
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: French/France (fr_FR)
```

Here are logs from /var/log/syslog :
```syslog
Aug 10 10:20:19 athena kernel: [ 3113.181635] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:1:85df3efd, in FreeCADLink [4808]
Aug 10 10:20:19 athena kernel: [ 3113.182656] i915 0000:00:02.0: [drm] Resetting rcs0 for stopped heartbeat on rcs0
Aug 10 10:20:19 athena kernel: [ 3113.183286] i915 0000:00:02.0: [drm] Resetting chip for stopped heartbeat on rcs0
Aug 10 10:20:19 athena systemd[1]: Starting autorandr execution hook...
Aug 10 10:20:19 athena kernel: [ 3113.285382] [drm:__uc_sanitize [i915]] *ERROR* Failed to reset GuC, ret = -110
Aug 10 10:20:19 athena kernel: [ 3113.386413] i915 0000:00:02.0: [drm] *ERROR* Failed to reset chip
Aug 10 10:20:19 athena kernel: [ 3113.386416] i915 0000:00:02.0: [drm:add_taint_for_CI [i915]] CI tainted:0x9 by intel_gt_reset+0x260/0x300 [i915]
Aug 10 10:20:19 athena kernel: [ 3113.489388] [drm:__uc_sanitize [i915]] *ERROR* Failed to reset GuC, ret = -110
Aug 10 10:20:19 athena kernel: [ 3113.491031] i915 0000:00:02.0: [drm] FreeCADLink[4808] context reset due to GPU hang
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE)
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) Backtrace:
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 0: /usr/lib/xorg/Xorg (OsLookupColor+0x13c) [0x55cecd857ddc]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 1: /lib/x86_64-linux-gnu/libpthread.so.0 (funlockfile+0x60) [0x7f1bd18c1420]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 2: /lib/x86_64-linux-gnu/libc.so.6 (gsignal+0xcb) [0x7f1bd16fc00b]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 3: /lib/x86_64-linux-gnu/libc.so.6 (abort+0x12b) [0x7f1bd16db859]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) unw_get_proc_name failed: no unwind info found [-10]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 4: /usr/lib/x86_64-linux-gnu/dri/iris_dri.so (?+0x0) [0x7f1bcf499498]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 5: /usr/lib/x86_64-linux-gnu/dri/iris_dri.so (nouveau_drm_screen_create+0x364e68) [0x7f1bd0185998]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 6: /usr/lib/x86_64-linux-gnu/dri/iris_dri.so (__driDriverGetExtensions_d3d12+0x79c58) [0x7f1bcf51e2c8]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 7: /usr/lib/xorg/modules/libglamoregl.so (glamor_destroy_pixmap+0x150) [0x7f1bd10723b0]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) unw_get_proc_name failed: no unwind info found [-10]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 8: /usr/lib/xorg/modules/drivers/modesetting_drv.so (?+0x0) [0x7f1bd2217b07]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 9: /usr/lib/xorg/Xorg (BlockHandler+0xa5) [0x55cecd6faa75]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 10: /usr/lib/xorg/Xorg (WaitForSomething+0x122) [0x55cecd851582]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 11: /usr/lib/xorg/Xorg (SendErrorToClient+0x117) [0x55cecd6f5d77]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 12: /usr/lib/xorg/Xorg (InitFonts+0x3b4) [0x55cecd6fa064]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 13: /lib/x86_64-linux-gnu/libc.so.6 (__libc_start_main+0xf3) [0x7f1bd16dd083]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) 14: /usr/lib/xorg/Xorg (_start+0x2e) [0x55cecd6e3aae]
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE)
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE)
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: Fatal server error:
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) Caught signal 6 (Aborted). Server aborting
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE)
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE)
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: Please consult the The X.Org Foundation support
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: #011 at http://wiki.x.org
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]:  for help.
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) Please also check the log file at "/home/dbozec/.local/share/xorg/Xorg.0.log" for additional information.
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE)
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (II) AIGLX: Suspending AIGLX clients for VT switch
Aug 10 10:20:19 athena autorandr[6822]: Can't open display :0
Aug 10 10:20:19 athena autorandr[6820]: Failed to run xrandr (line 480)
Aug 10 10:20:19 athena autorandr[6820]: autorandr running as user dbozec (started from batch instance)
Aug 10 10:20:19 athena autorandr[6808]: Running autorandr as dbozec for display :0
Aug 10 10:20:19 athena systemd[1]: autorandr.service: Succeeded.
Aug 10 10:20:19 athena systemd[1]: Finished autorandr execution hook.
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[2262]: (EE) Server terminated with error (1). Closing log file.
Aug 10 10:20:19 athena pulseaudio[2164]: X connection to :0 broken (explicit kill or server shutdown).
Aug 10 10:20:19 athena nm-applet[2916]: nm-applet: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
Aug 10 10:20:19 athena firefox[3083]: firefox: Fatal IO error 11 (Resource temporarily unavailable) on X server :0.
[... 14 lines cut ...]
Aug 10 10:20:19 athena dbus-daemon[2171]: [session uid=1000 pid=2171] Activating via systemd: service name='org.freedesktop.impl.portal.desktop.gtk' unit='xdg-desktop-portal-gtk.service' requested by ':1.31' (uid=1000 pid=2683 comm="/usr/libexec/xdg-desktop-portal " label="unconfined")
Aug 10 10:20:19 athena systemd[2152]: Starting Portal service (GTK+/GNOME implementation)...
Aug 10 10:20:19 athena xdg-desktop-portal-gtk[6832]: Unable to init server: Could not connect: Connection refused
Aug 10 10:20:19 athena xdg-desktop-por[6832]: cannot open display: :0
Aug 10 10:20:19 athena systemd[2152]: xdg-desktop-portal-gtk.service: Main process exited, code=exited, status=1/FAILURE
Aug 10 10:20:19 athena systemd[2152]: xdg-desktop-portal-gtk.service: Failed with result 'exit-code'.
Aug 10 10:20:19 athena systemd[2152]: Failed to start Portal service (GTK+/GNOME implementation).
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[4799]: /tmp/.mount_freecakACpf5/AppRun: line 46:  4808 Aborted                 "$PREFIX/bin/$name" "$@"
Aug 10 10:20:19 athena systemd[1]: tmp-.mount_joplinOTAaRT.mount: Succeeded.
Aug 10 10:20:19 athena systemd[2152]: tmp-.mount_joplinOTAaRT.mount: Succeeded.
Aug 10 10:20:19 athena systemd[2152]: tmp-.mount_freecakACpf5.mount: Succeeded.
Aug 10 10:20:19 athena systemd[1]: tmp-.mount_freecakACpf5.mount: Succeeded.
Aug 10 10:20:19 athena spacenavd[1118]: already using device: 3Dconnexion 3Dconnexion Universal Receiver (/dev/input/event11)
Aug 10 10:20:19 athena spacenavd[1118]: failed to find any supported devices
Aug 10 10:20:19 athena spacenavd[1118]: already using device: 3Dconnexion 3Dconnexion Universal Receiver (/dev/input/event11)
Aug 10 10:20:19 athena spacenavd[1118]: failed to find any supported devices
Aug 10 10:20:19 athena spacenavd[1118]: already using device: 3Dconnexion 3Dconnexion Universal Receiver (/dev/input/event11)
Aug 10 10:20:19 athena spacenavd[1118]: failed to find any supported devices
Aug 10 10:20:19 athena spacenavd[1118]: already using device: 3Dconnexion 3Dconnexion Universal Receiver (/dev/input/event11)
Aug 10 10:20:19 athena spacenavd[1118]: failed to find any supported devices
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[6660]: Exiting due to channel error.
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[3474]: Exiting due to channel error.
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[6624]: Exiting due to channel error.
Aug 10 10:20:19 athena /usr/lib/gdm3/gdm-x-session[3528]: Exiting due to channel error.
```

dbozec, 2022-08-11:
Happened again few minutes ago.
I'm switching back to FreeCAD-asm3-Stable-Conda-Py3-Qt5-20220430-glibc2.12-x86_64.AppImage

luzpaz, 2022-08-12:
Can you reproduce on the Daily branch ?

dbozec, 2022-08-12:
Yes, happened again :
```syslog
Aug 12 19:39:28 athena kernel: [39287.356811] i915 0000:00:02.0: [drm] GPU HANG: ecode 12:1:85dffffb, in FreeCADLink [46402]
Aug 12 19:39:28 athena kernel: [39287.357832] i915 0000:00:02.0: [drm] Resetting rcs0 for stopped heartbeat on rcs0
```
```
OS: Ubuntu 20.04.4 LTS (openbox)
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: French/France (fr_FR)
```

realthunder, 2022-08-19:
Probably a graphics driver bug. Just a wild guess, https://bugs.mageia.org/show_bug.cgi?id=27159

dbozec, 2022-08-19:
I just updated my Ubuntu to 22.04, will see if it happens again