realthunder/FreeCAD_assembly3 #1261: Hidding bodies is not working within assemblies

not-a-number-dot-org, 2022-08-14:
'visiblity' property set to false/true/expression is working on bodies outside of assemblies. But if the body is part of an assembly it doesn't work.

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.709.28846 (Git)
Build type: Release
Branch: LinkStage3
Hash: c880c050c5f7601e4bae6740de71500edf2be0c7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: French/France (fr_FR)

realthunder, 2022-08-19:
The visibility of the a child object in the `Parts` group is controlled by the `Parts.VisibilityList` property. You can use the `setElementVisible()` API to control the child visibility, e.g. `Parts.setElementVisible('ChildObjectName', False)`.

not-a-number-dot-org, 2022-08-21:
In the view tab there is a "visibility" property for assemblies which is only visible if the display of all properties has been requested.
Wouldn't it be a good idea to make this property visible by default?
I still hope to be able to create conditional assemblies (see #967).
We can now suppress features and associated elements, and disable constraints. Looks like hiding elements that are not required in a conditional assembly is the final step.

realthunder, 2022-08-21:
Expression engine should be able to do what you requested here. It would be better to post an example assembly here for further discussion.

not-a-number-dot-org, 2022-08-27:
Here is an example.
Body data property Visibility is set to an expression that resolve to false.
When the Body is added to an assembly, it becomes visible even if Visibility is still false.

Maybe Visibility property is not the right property I need for conditional assembly.
I guess I'm rather looking for a Suppress property at Body level or even better at Assembly level.

Something that would allow to hide an assembly and disable its associated constraints with one expression.

Assume I have a crossover on which I need to make N mortises and assemble N uprights based on the number of drawers. Actually I can suppress unnecessary features for mortises and disabled their elements and constraints not needed but I can't suppress/hide unnecessary uprights.
I can of course set Visibility property of the mortises assemblies in the View tab but wouldn't it be better to have a property at the data level for hiding unnecessary uprights?
 
[hidden body.FCStd.gz](https://github.com/realthunder/FreeCAD_assembly3/files/9437602/hidden.body.FCStd.gz)

not-a-number-dot-org, 2022-08-28:
Well... it doesn't work at all.
I have my [model](https://forum.freecadweb.org/viewtopic.php?f=24&t=71459) done.
I have two variables in a spreadsheet to configure the built; one for the external veneers and one for the internal. Suppress/modifications of features/constraints based on the variables works fine but I can't find a way to hide the veneers not needed.
Either setting Assembly Data->Visibility or View->Visibility to an expression doesn't hide the veneers not needed; they have to be hided manually.

not-a-number-dot-org, 2023-10-14:
Hi again.

Here is another example :
Body.Visibility, Body001.Visibility and Assembly001 are all set to <<ss>>.visibility

If <<ss>>.visibility is set to 0
Body and Assembly001 are hidden but not Body001

[visibility.zip](https://github.com/realthunder/FreeCAD_assembly3/files/12904189/visibility.zip)

realthunder, 2024-02-03:
Hi, sorry for the very late reply. Somehow missed your post. It is not easy to control child visibility of an assembly 3 container which is based on `App::LinkGroup`. Right now, you can only do this through manually entering command in console. For example,

```
App.ActiveDocument.Parts.setExpression('VisibilityList', '[<<ss>>.visibility]')
```

This sets an expression for the `VisibilityList` property on `Parts` in `Assembly`. The expression should evaluate to a list of boolean values, with each entry corresponding to the visibility setting of a child.