realthunder/FreeCAD_assembly3 #1264: Constraint error is not clear

ceremcem, 2022-08-22:
I removed an object and its elements/constraints as far as I could. After many successful progress on my model, there is suddenly a "constraint related" error, which is unclear to me: 

```
<class 'TypeError'>: argument 1 must be App.DocumentObject, not None
14:52:20  218.775 <App> Document.cpp(4293): Failed to recompute housing2#Assembly: argument 1 must be App.DocumentObject, not None
14:52:20  Recompute failed! Please check report view.
```

There is no error icon in front of any of the constraints. How can I detect where the erroneous constraint (if there is any)?

ceremcem, 2022-08-23:
I disabled the constraints one by one and the error is resolved. Apparently the erroneous constraints/elements are not shown correctly. 

(I'm sending my model privately. You'll see the error when you open the project. See that `_Element033` is a broken element inside `PlaneAlignment013`. 

Also `PlaneAlignment014` is erroneous. When you disable ..13 and ..14, solver "compiles" correctly. If you enable ..14, solver fails, the parent assembly container icon shows an error message but ..14's icon does not show any error icon.)

mgth, 2022-10-25:
Had the same problem, In my assembly, first part was fixed, then five parts plane aligned, averything was ok.
But adding a sixth I had the same error without adding any constraint.
And removing the part did not fix it.
But disabling the fixed constraint resolved the problem.
I thought it was good practice to have at least one part fixed.