realthunder/FreeCAD_assembly3 #1271: We need a helper plane tool for centering parts only vertically in between two points

ceremcem, 2022-08-24:
[image]

Here I want to create a constraint that will make `d1` and `d2` equal, while letting me move this part freely in "up/down" direction. 

This is the same requirement we discussed in #53. We need a convenient way to achieve this kind of positioning tasks. 

My proposal is having a tool that creates a `Helper Plane` in between two points. The invisible line between two points will be perpendicular to this helper plane.

jpg87Fr, 2022-08-27:
I would declare the socket's symmetry plane as an element, then put a symmetry constraint between a and b with respect to that plane.

ceremcem, 2022-08-29:
@jpg87Fr Yes, that would solve the problem but you have to define a symmetry plane for every part. It would be the same or more burden for the parts we just download from manufacturer and import into the model (the STL files). This may also be preferred, it's just another approach.