realthunder/FreeCAD_assembly3 #1278: "Infinite shape provided"

Erhannis, 2022-10-11:
Not sure whether to ask about this on the forum or here, but it seems more polite to start here than ask about a fork on the main forum.
Here's a file.
[staple_bracket.zip](https://github.com/realthunder/FreeCAD_assembly3/files/9749959/staple_bracket.zip)
I tried to create a pretty simple shape - made a new file, body, part, sketch, and a polyline in a wall bracket shape.  Added constraints.  Left sketch mode, and didn't see anything - the sketch item said "AttachEngine::getInertialPropsOfShape: infinite shape provided".  Any idea what's up with that?  I only get 4 google results, and none of them seemed helpful.

Erhannis, 2022-10-11:
I tried doing approximately the same thing again, and it worked fine.  I remember originally trying some constraints that didn't do what I expected, and removed them.

realthunder, 2022-10-12:
The message is caused by invalid selection of the sketch attachment selection. Maybe you accidentally changed it. Just right click the sketch, and choose `Attachment` to manually correct it.

Erhannis, 2022-10-12:
That seems to have done it; thanks.  I went through several weird states but by comparing with a good version, managed to get it back to correct.  (I also somehow had the sketch in construction mode.)  Thanks!

closed 2022-10-12 by Erhannis