realthunder/FreeCAD_assembly3 #1281: missing color preferences

gneiss15, 2022-10-22:
I foud this branch via: 
Design a Rasberry Pi case for 3d printing tutorial using FreeCAD LinkStage3 [https://www.youtube.com/watch?v=OWNrYvxpG4k]
[https://www.youtube.com/watch?v=OWNrYvxpG4k](url)

realthunder, 2022-10-23:
Your quoted link on interface customization is a bit outdated. You can still follow it, but there is much easier way to switch UI, through `Tools -> Preset configurations`. The most up-to-date tutorial video I can find is [here](https://youtu.be/i-Dh6zC7DlI).

Also, new version of my build is released [here](https://github.com/realthunder/FreeCAD/releases). I encourage you to download `Daily` build instead of `Stable` one. I plan to switch the current `Daily` branch to `Stable` soon.

As for the text color settings you mentioned, you can change the text color of the `Python console` is in `Preferences -> General -> Editor`. For `Report view`, it's in `Preferences -> General ->  Output window`.

gneiss15, 2022-10-23:
Thanks for Your explanations.
Especialy for the locations to change the colors...
I will watch the new customization video ASAP.

> new version of my build 
That's the one I already use (since yesterday ;-)

Maybe You can explain me someting that I don't understand and didn't find any explanation via web-search:
When customizing keyboard there is a "Shortcut priority list".
What's this ?
What happens if more than one action is bound to the same key(-combination) ?
What is the priority(-role) ?

BTW: There is one typo inside the "customize" window "Shorcut" instead of "Shortcut":
[image]

realthunder, 2022-10-23:
Thanks for the heads up for the typo.

The common case of assigning the same shortcut to different commands is when those commands appear in different workbench, which means they won't appear at the same time. Just think of those toolbar buttons,. When you switch workbench, some toolbars are hidden, while some are shown.

In case those commands with the same shortcut do appear at the same time, the priority list is consulted to decide which command to run.

gneiss15, 2022-10-23:
Ah, OK, so it works as expected (same key different WB)..
If You use the same key inside the same WB (or Global) the second prio one would never be executet, right ?
I hope I get time tomorro for watching the video & checking the colors.. I will report back if something ist still missing/unclear.

Thanks for being very helpy

gneiss15, 2022-10-24:
I just read Your prev answer in more detail...
You mentioned how to change the **text color** of the Python console, but not how to change the **background** color(s) (of all the bottom windows).
The problem is, that with the given sytles yellow or red text colors (with are good for warnings / error) look bad in a "dark-blue" background.