realthunder/FreeCAD_assembly3 #1282: Ubuntu 18.04 install with current AppImage FreeCAD_0.21.29457-2022-10-22-conda-Linux-x86_64-py310.AppImage failed

CBtiger, 2022-10-24:
Not sure yet what the problem is. As reported on the Freecad issues [https://github.com/FreeCAD/FreeCAD-Bundle/issues/139], the Asm3 update failed in my working setup and a new fresh setup on Ubuntu 18.04. 

I tried with both builds with clean user settings to install Asm3 via AddOnManager. Both times Asm3 installed but did not work. 
FreeCAD_0.21.29457-2022-10-22-conda-Linux-x86_64-py310.AppImage
FreeCAD_weekly-builds-28611-Linux-Conda_glibc2.12-x86_64.AppImage

The "Assembly" icon in the tree was missing
I ran a simple test scene with cube and cylinder and 1 Lock constraint and 1 Plane Alignment constraint. 
The two left "Solve" buttons did not toggle state on the toolbar any more. but they did something as the console. It showed
```
10:19:18  48.891045 <asm3.sys> solver.py(412): no assembly found
10:19:22  4.087824 <asm3.sys> solver.py(412): no assembly found
```

During setup (I had to install twice) py-slvs did install.
```
10:22:22  Looking in indexes: http://localhost:3141/root/pypi/+simple/
Collecting py-slvs
  Downloading http://localhost:3141/root/pypi/%2Bf/f51/3627031c5da9d/py_slvs-1.0.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (294 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 294.1/294.1 kB 374.5 MB/s eta 0:00:00
Installing collected packages: py-slvs
Successfully installed py-slvs-1.0.3
```

Can someone please confirm that asm3 could be installed with any of those two AppImages? 

PS: is there any way to manually install py_slvs to the AppImage, I mean without AddonManager?  I am using pyenvs on my computers and I would rather not like to install py_slvs into my global environment.

CBtiger, 2022-10-24:
AddonInfo: I suspeced the problem might have to do with  _py_slvs_. The AddOn Manager installed the _py_slvs_ package into the following path
```
 ~/.local/lib/python3.6/site-packages/py_slvs ~/
    __init__.py
    slvs.py
    _slvs.so 
```
In the Python Console in FreeCAD  "import slvs" failed, but "from py_slvs import slvs" worked.

After I copied the above files to the asm3 folder as below (from the info about manual install). Now the "import slvs" worked as well the Python Console. 
```
  ~/.local/FreeCAD/Mod/Assembly3/
       __init__.py
       slvs.py
       _slvs.so 
       package.xml
       freeccad
              asm3/
                   <other asm3 files>
```
In the file _sys_slvs.py_ I saw that both ways should work. But Asm3 still showed the same result both in 0.20 and 0.21 packages.

Attached the testfile used
[Asm3Test.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/9850830/Asm3Test.FCStd.zip)

realthunder, 2022-10-25:
> During setup (I had to install twice) py-slvs did install.

How did you install it twice?

> ~/.local/lib/python3.6/site-packages/py_slvs ~/
>     __init__.py
>     slvs.py
>     _slvs.so

Where did you get this path? FreeCAD AppImage uses newer Python, so it most definitely will not use this path.

After some digging, I found out that the new AddonManager installs the dependency in `.local/share/FreeCAD/AdditionalPythonPackages` which may cause problem if you are alternating AppImages that pack  different version of Python, because `py_slvs` is a native package. Please go in this directory and manually remove all `py_slvs` directories. Then start FreeCAD, and if you've already have `asm3` installed, switch to the workbench and it will prompt to install `py_slvs`. Try and see if this can get it installed. If you haven't installed `asm3` (or removed it previously), when prompt to install dependency, choose `Ignore` and try.

CBtiger, 2022-10-25:
> How did you install it twice?

I installed it with AppManager, but got no python prompt. Then I uninstalled it and installed it again. Then I got the AddOnManager's pyhton install prompt and installed that too.

> 
> > ~/.local/lib/python3.6/site-packages/py_slvs ~/
> Where did you get this path? FreeCAD AppImage uses newer Python, so it most definitely will not use this path.

I found the path on my computer. But you are right, there is also a Python3.10 path beside it: `  ~/.local/lib/python3.10/site-packages/py_slvs ~/`. So I have two paths, I guess from different Asm3 versions. 

That the path is created by your WB. As you say the AddOnManager installs to AdditionalPythonPackages. It does not remove those installs if the WB is uninstalled. Maybe that is not intended.

> Please go in this directory and manually remove all `py_slvs` directories. Then start FreeCAD, and if you've already have `asm3` installed, switch to the workbench and it will prompt to install [...]

OK. I removed everything and did as you say. I got the WB install prompt when I opened the WB. It created the path   `~/.local/lib/python3.10/site-packages/py_slvs ~`/ again and gave the output below in the Report View. 

```
11:01:38  1.275895 <asm3.main> init_gui.py(14): no solver backend found
11:02:51  /tmp/.mount_FreeCAkDbD3w/usr/bin/python -m pip install py-slvs --user
11:02:51  Looking in indexes: http://localhost:3141/root/pypi/+simple/
Collecting py-slvs
  Downloading http://localhost:3141/root/pypi/%2Bf/f51/3627031c5da9d/py_slvs-1.0.3-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (294 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 294.1/294.1 kB 388.0 MB/s eta 0:00:00
Installing collected packages: py-slvs
Successfully installed py-slvs-1.0.3
```

Then it showed me the dialog "Failed" "Failed to install py-slvs". When I loaded my test scene, the WB highlighted my constraint surfaces, so the basic functions work. The tree Assembly icon is still missing. When I tried to solve the scene it did nothing exceot the output below. 

```
>>> Gui.runCommand('asm3CmdSolve',0)                                # python console

11:16:08  870.453669 <asm3.sys> solver.py(412): no assembly found   # Report View

```

realthunder, 2022-10-25:
> Then it showed me the dialog "Failed" "Failed to install py-slvs".

Can you restart FreeCAD, and try again? When you see this `Failed` dailog but the report view shows success, just restart FreeCAD and see if it works.

CBtiger, 2022-10-25:
Nope, after Restart still the same situation. 

`16:25:55  1.015991 <asm3.sys> solver.py(412): no assembly found`

Not sure if relevant, but I noted the following from the console. So the lib seems to be  there. 
```
>> import slvs                                         # failed
>> from py_slvs import slvs                    # fworked

```

realthunder, 2022-10-26:
It is probably already working. The module is named `py_slvs`, so if `import py_slvs` works then it should be ok. You didn't see the `no solver found` warning right? Try create a new assembly and see. You're probably trying an assembly saved when there was no solver. If so, you'll need to manually set the `Assembly.SolverType` property to `SolveSpace`.

CBtiger, 2022-10-26:
Excellent. That worked.
The Assemly/ Property SolverType=None also caused the missing Icon in the Tree. After selecting the only option "SolverSpace" the icon was back
I was not aware that the file saves the solver type. I will note that behaviour for the message "no assembly found". 

Maybe its an idea to improve the error message "no assembly found" -> "no assembly found or no solver type set." which would be a bit more intuitive

luzpaz, 2022-10-26:
Maybe we need to rename ticket to "Make Assembly3 install `py_slvs` via the modernized Addon Manager"

CBtiger, 2022-10-26:
> Maybe we need to rename ticket to "Make Assembly3 install `py_slvs` via the modernized Addon Manager"
done

CBtiger, 2022-10-30:
opened a ticket to remove the AddOnManager install fuction for py-slvs

closed 2022-10-30 by CBtiger