realthunder/FreeCAD_assembly3 #1283: M key on polyline tool not working

eefweenink, 2022-10-26:
After installing and using touchpad (on macbook) the polyline (part design) does not react as exspected. Nothing happens when clicking the M. 

Any solutions, workaround? 

Regards, Eef

eefweenink, 2022-10-26:
NB : I could solve it this way, but don't know if this effects other stuff. 
Tools -> Customize -> Keyboard -> search for polyline
set shortcut to m

realthunder, 2022-10-26:
You solution is the correct thing to do as the upstream Sketcher workbench has changed default 'm' shortcut along with many other commands.

alber70g, 2023-05-28:
Still doesn't work for me. Just downloaded the most recent version (weekly) but for me hitting "M" for the second time just disables polyline... :( Other ideas?

raindropsfromsky, 2023-06-21:
I am using Realthunder FreeCAD 2023.524.0.14555 on Windows 11 Home.
I face the same issue.

I checked out the setting, and it seems to be correct: 
[image]

But there is only one keyboard shortcut related to polyline: Only for creating a new polyline.
There is no separate entry in this list to allow us to set a different shortcut to cycle through the modes.

raindropsfromsky, 2023-06-21:
Update:

The shortcut SHFT+M works.

But there are three problems (bugs):
1. The Customize GUI does not show this shortcut at all. 
2. It is not possible to change this shortcut to something else. 
3. The SHFT+M shortcut requires us to use both hands. But we can use only one hand because the other hand is busy in drawing the polyline! In other words, a single-key shortcut is needed.