realthunder/FreeCAD_assembly3 #1284: Polilyne in sketch no cycle behaviour with m key.

GKSeifert, 2022-10-27:
Freecad Link Branch
2022.1026
Rev number 14555 (Got shallow)
Operating system Ubuntu Core 20
Word size 65 bit
Branch   (HEAD detached at 2022.10.26-edge)
Installed via SNAP

M does not (alone) resulting the required cyclic behavior.  GM does initiate polyline so the shortcut has not been changed. GM will result in the cyclic behavior but that is NOT the way it is supposed to work and is not an acceptable substitute.

realthunder, 2022-10-27:
The decision of changing the default shortcuts is made by the developers of Sketcher workbench.

GKSeifert, 2022-10-28:
In spite of hundreds of tutorials, books, videos and so on that lead one to expect M alone to result in the cyclic action? I don't believe the shortcut has been changed. Freecad 20.1 on Linux, both flatpak and appimage do work as documented. M alone works. The same Sketcher I would think.

GKSeifert, 2022-10-28:
I just noticed #1283. I decided to try changing the default shortcut (G,M) to M. Tools, Customise, Keyboard, Category shows no Category for Sketcher at all. Searching for any of the commands I find in Freecad 20.1's Keyboard Customization, under Sketcher, is unsuccessful. It seems all of the Sketcher shortcuts have gotten lost, at least partially. G,M does initiate the polyline tool, and also in the cyclic action. But it seems they are not being handled normally.   Also, I have no idea if Sketcher in this product is the same as in Freecad 20.1. Sorry about that assumption.

GKSeifert, 2022-10-28:
Ha! That didn't take long. Restarted it and it came up with today's date, and all the Sketcher shortcuts in Keyboard Customization too.  M now works after changing the shortcut. Works for me. Thanks!