realthunder/FreeCAD_assembly3 #1286: Cannot create new element in frozen assembly

abetis, 2022-10-30:
RuntimeError: Cannot create new element in frozen assembly

Flow:
Create an assembly, add link to a sub-assembly.
Freeze the sub-assembly.
There is a need to add another link of the same sub-assembly to the main assembly.
You'll get the RuntimeError: Cannot create new element in frozen assembly

Need to open the sub-assembly file, unfreeze it and only then could add the link to the sub-assembly.

As far as I see it, the element is added in the main assembly, and not in the sub-assembly.
Am I wrong?

Why sub-assembly document have to be touched to add it to the main assembly?
I don't try to change it, all I want is use the existing model in bigger assembly.

Also, it will be more convenient to have a way to change the freeze flag from the main assembly link entry in the tree, without the need to open each document I want to add again to unfreeze and after the addition to freeze it again, since performance issues require me to do that.

luzpaz, 2022-11-08:
Please ask on assembly subforum https://forum.freecadweb.org/viewforum.php?f=20

abetis, 2022-11-08:
Ask what?
I see current behavior as an issue.
If you freeze a sub-assembly after you finish its development, or after adding it in a larger assembly and want to reduce calculations overhead... it's not logical to unfreeze it only for adding another instance to the large assembly.
Especially when you don't intend to do any change in the sub-assembly, all you want to do is add another instance.

abetis, 2022-11-08:
I guess #1289 is related to that issue, since in both cases frozen state prevents normal logical operation, although the frozen sub-assembly is not changed.

realthunder, 2022-11-09:
The element in the main assembly is linked to its child part. If that part is a sub-assembly, then it is forced to link to an `Element` in the sub-assembly. This is an important concept in assembly3. You can consider the `Elements` of an assembly as its interface in order to abstract its internal parts/sub-assemblies from higher level assembly. So, when you model an assembly, try to think before hand which elements your want to export to be used by higher level assembly.

abetis, 2022-11-09:
It's possible for parts that you create by yourself from the ground and prepare those interface points, but not for a pre-manufactured parts with thousands of the interfaces that can be used as an attachment point in the sub-assembly.

I understand the limitation from the design point of view. 
Just saying that the user experiance from the user point of view is not good.
Trying to do a measurement for example, as in #1289 is a good example for that.