realthunder/FreeCAD_assembly3 #1287: Sometimes FreeCAD makes wrong intersecting geometry in PartDesign workbench

AlexStankov, 2022-10-31:
OS: Ubuntu 20.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

Steps to reproduce:
1. Extrude cylinder in part design workbench:
[image]
2. Extrude another cylinder in part design workbench which intersects first one. Preview looks good:
[image]
But when I press OK button - FreeCAD shows strange intersction:
[image]

Expected behavior: FreeCAD adds second cylinder to the first one.
Observed behavior: FreeCAD creates strange form which is not a solid body.

This behavior have been checked in two latest FreeCAD versions: 22.07.09 and 22.10.21.

AlexStankov, 2022-10-31:
Project source:
[hanger1.zip](https://github.com/realthunder/FreeCAD_assembly3/files/9899315/hanger1.zip)

AlexStankov, 2022-10-31:
OS: Ubuntu 20.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 2022.1021.26244 +5249 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: d307b5c936403e8dca3f24b5c1e64bd16c88b28d
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

With latest release result is the same:
[image]

realthunder, 2022-10-31:
That's strange. I can't reproduce the problem using the exact same AppImage? Just right click the second Pad, and choose 'Recompute' gives the correct output. Can you please post your settings here, `Tools -> Edit parameters -> Export`.

AlexStankov, 2022-10-31:
Here is the settings export:
[FreeCADSettings.zip](https://github.com/realthunder/FreeCAD_assembly3/files/9903313/FreeCADSettings.zip)

AlexStankov, 2022-10-31:
I've tried to recompute, but result was the same.