realthunder/FreeCAD_assembly3 #1289: Can't measure distance between frozen assemblies

abetis, 2022-11-04:
Trying to use measurement tools that involve a frozen assembly part/face will give an error: 
Cannot create new element in frozen assembly

luzpaz, 2022-11-06:
Missing:  
* FreeCAD [About](https://wiki.freecad.org/About) version  
* Example file
* Example screenshot

abetis, 2022-11-06:
Sorry, always forget about it. Attaching ZIP with 3 models, a shaft and 2 links to frozen sub-assembly.

OS: Ubuntu 20.04.5 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.1.29410 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.1)
Hash: f5d13554ecc7a456fb6e970568ae5c74ba727563
Python 3.10.5, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.4
  * Assembly3 0.11.3

[test-distance.zip](https://github.com/realthunder/FreeCAD_assembly3/files/9946217/test-distance.zip)

[image]

jpg87Fr, 2022-11-09:
I tested the PointsPlaneDistance constraint from the elements you had defined and then the MeasurePointPlane tool and had no problems. It is obvious that these elements must have been defined before freezing your assemblies, otherwise they must be temporarily unfrozen.
[image]

abetis, 2022-11-09:
Using existing elements to measure between them obviously won't need to create new elements and hence there will be no need to unfreeze the sub-assembly.
That will follow the application design concept, but not a good user experience in my opinion.
You can't really know ahead what elements you might need before freezing the sub-assembly.

How do you guys use the measurement tools?
You keep them in the assembly or delete right after you see the value?

jpg87Fr, 2022-11-09:
Of course we can't anticipate everything in a design process, but for certain industrial components, the functional elements are often the same, and for the elements being designed... I don't think we can escape the 'unexpected, so unfreeze the subassembly, modify, and refreeze.

As for the results of the measurement tools, we may need to keep the memory of them for a while, but in general I delete them after having read the information.

abetis, 2022-11-10:
I understand the point.
Just saying that it's not a good user-experience.

The freeze functionnality as I see it was meant to help with the performance, maybe we should solve it so there will be no need to freeze sub-assemblies consisting of 2-3 parts...
#1285

jpg87Fr, 2022-11-10:
For me, freezing is not systematic. Only if my system is struggling.

abetis, 2022-11-20:
Mine is struggling... Especiallly for some round parts, that's the only reason I use freeze and I guess that's the reason why it was added.