realthunder/FreeCAD_assembly3 #1293: FC crashing on adding a Arch / Dimension

mjollnir14, 2022-12-19:
Reproduce:

- Right-click the file name in the tree view, and create a new Group. Right click the new Group and rename to: Plan dimensions and repeat using the new name Elevation dimensions.
- Set the [Working Plane](http://www.freecadweb.org/wiki/index.php?title=Draft_SelectPlane) to XY plane
- Make sure the icon [restrict](http://www.freecadweb.org/wiki/index.php?title=Draft_WorkingPlane) snap location is turned on, so everything you draw stays on the working plane.
- Draw a [dimensions](http://www.freecadweb.org/wiki/index.php?title=Draft_Dimension), for example as on the image below.

> OS: Ubuntu Core 20 (ubuntu:GNOME/ubuntu)
> Word size of FreeCAD: 64-bit
> Version: 2022.1213.14555 (Git shallow)
> Build type: Release
> Branch: (HEAD detached at 2022.12.13-edge)
> Hash: d033846145cd695a3aac53cfbaab0f9bf198702c
> Python version: 3.8.10
> Qt version: 5.15.5
> Coin version: 4.0.1
> OCC version: 7.6.3
> Locale: English/United States (en_US)

luzpaz, 2022-12-19:
Is this Assembly3 related or Link branch related ?

mjollnir14, 2022-12-20:
Linkbranch I guess