realthunder/FreeCAD_assembly3 #1303: adding part to group of parts, makes freecad crash

eefweenink, 2023-03-18:
OS: macOS 11.7
Word size of FreeCAD: 64-bit
Version: 2023.131.26244 +5365 (Git)
Build type: Release
Branch: LinkDaily
Hash: 73c4ca254c5de8849d104973482e98399fb24863
Python version: 3.9.10
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: Dutch/Netherlands (nl_NL)

I use groups to quickly hide/unhide several objects at the same time. and also to keep some order in the list of objects. 
works great. Now I have this issue: 
- when adding a new object directly into a group it sometimes (Not always) crashes freecad. 
after restart, it stays the same. 
Only way to solve this: 
-  delete the existing group; add new part to parts and create new group with all items. 
regards, Eef

realthunder, 2023-03-19:
Can you please check MacOS crash report and post the call stack here?

eefweenink, 2023-03-19:
Process:               FreeCADLink [85474]
Path:                  /Applications/FreeCAD.app/Contents/MacOS/FreeCADLink
Identifier:            ???
Version:               ??? (20230131)
Code Type:             X86-64 (Native)
Parent Process:        ??? [85467]
User ID:               501

Date/Time:             2023-03-18 20:59:46.756 +0100
OS Version:            macOS 11.7.4 (20G1120)
Report Version:        12
Anonymous UUID:        B8F2EA0F-9CD5-6605-054A-5E30E0583135

Sleep/Wake UUID:       618DCE0D-866A-48AB-A381-83BC22BD36FE

Time Awake Since Boot: 190000 seconds
Time Since Wake:       190000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [85474]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libCoin.80.dylib              	0x000000010a4d19d4 SoSeparatorP::invalidate_gl_cache(void*, void*) + 4
1   libCoin.80.dylib              	0x000000010a2990dd cc_dict_apply + 93
2   libCoin.80.dylib              	0x000000010a6468c1 cc_storage_apply_to_all + 49
3   libCoin.80.dylib              	0x000000010a4d17af SoSeparator::notify(SoNotList*) + 63
4   libCoin.80.dylib              	0x000000010a45e892 SoBase::startNotify() + 82
5   libCoin.80.dylib              	0x000000010a4618fc SoChildList::insert(SoNode*, int) + 76
6   libCoin.80.dylib              	0x000000010a4b4413 SoGroup::insertChild(SoNode*, int) + 67
7   libFreeCADGui.dylib           	0x0000000108cf5a24 Gui::LinkView::setChildren(std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*> > const&, boost::dynamic_bitset<unsigned long, std::__1::allocator<unsigned long> > const&, Gui::LinkView::SnapshotType) + 4180
8   libFreeCADGui.dylib           	0x0000000108cffd10 Gui::ViewProviderLink::updateElementList(App::LinkBaseExtension*) + 688
9   libFreeCADGui.dylib           	0x0000000108cfe107 Gui::ViewProviderLink::updateData(App::Property const*) + 103
10  libFreeCADGui.dylib           	0x0000000108ca8d15 Gui::ViewProviderDocumentObject::update(App::Property const*) + 165
11  libFreeCADGui.dylib           	0x000000010878c3bb Gui::Document::slotChangedObject(App::DocumentObject const&, App::Property const&) + 139
12  libFreeCADApp.dylib           	0x0000000109be48a9 boost::signals2::detail::slot_call_iterator_t<boost::signals2::detail::variadic_slot_invoker<boost::signals2::detail::void_type, App::DocumentObject const&, App::Property const&>, std::__1::__list_iterator<boost::shared_ptr<boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (App::DocumentObject const&, App::Property const&), boost::function<void (App::DocumentObject const&, App::Property const&)> >, boost::signals2::mutex> >, void*>, boost::signals2::detail::connection_body<std::__1::pair<boost::signals2::detail::slot_meta_group, boost::optional<int> >, boost::signals2::slot<void (App::DocumentObject const&, App::Property const&), boost::function<void (App::DocumentObject const&, App::Property const&)> >, boost::signals2::mutex> >::dereference() const + 73
13  libFreeCADApp.dylib           	0x0000000109be43d1 boost::signals2::detail::signal_impl<void (App::DocumentObject const&, App::Property const&), boost::signals2::optional_last_value<void>, int, std::__1::less<int>, boost::function<void (App::DocumentObject const&, App::Property const&)>, boost::function<void (boost::signals2::connection const&, App::DocumentObject const&, App::Property const&)>, boost::signals2::mutex>::operator()(App::DocumentObject const&, App::Property const&) + 689
14  libFreeCADApp.dylib           	0x0000000109c1ea2e App::DocumentObject::onChanged(App::Property const*) + 1838
15  Part.so                       	0x00000001136f2532 Part::Feature::onChanged(App::Property const*) + 274
16  Part.so                       	0x00000001136f811d App::FeaturePythonT<Part::Feature>::onChanged(App::Property const*) + 109
17  libFreeCADApp.dylib           	0x0000000109d80251 App::Property::touch() + 113
18  libFreeCADApp.dylib           	0x0000000109da6b51 App::PropertyLinkBase::hasSetValue() + 81
19  libFreeCADApp.dylib           	0x0000000109db1d7e App::PropertyListsT<App::DocumentObject*, std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*> >, App::PropertyLinkListBase>::setValues(std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*> >&&) + 238
20  libFreeCADApp.dylib           	0x0000000109db14ec App::PropertyLinkList::setValues(std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*> >&&) + 1036
21  libFreeCADApp.dylib           	0x0000000109c2ebe4 App::PropertyListsT<App::DocumentObject*, std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*> >, App::PropertyLinkListBase>::setValues(std::__1::vector<App::DocumentObject*, std::__1::allocator<App::DocumentObject*> > const&) + 132
22  libFreeCADApp.dylib           	0x0000000109d42fa8 App::LinkBaseExtension::updateGroup() + 2904
[... cut ...]
    0x7fff32a5a000 -     0x7fff32babfff  com.apple.CoreWiFi (3.0 - 341) <1018F0E7-2578-38AD-B47D-361E41457701> /System/Library/PrivateFrameworks/CoreWiFi.framework/Versions/A/CoreWiFi
    0x7fff32bac000 -     0x7fff32bc6fff  com.apple.BackBoardServices (1.0 - 1.0) <49254756-6605-3D38-B6BB-61B1F7DB46A8> /System/Library/PrivateFrameworks/BackBoardServices.framework/Versions/A/BackBoardServices
    0x7fff32bc7000 -     0x7fff32bfefff  com.apple.LDAPFramework (2.4.28 - 194.5) <7CB1BD1C-3A2D-3B02-9B21-F4ED9ABD12DD> /System/Library/Frameworks/LDA