realthunder/FreeCAD_assembly3 #1306: Additive helix deforms shapes

marcx95, 2023-04-22:
[image]
[image]
[image]
[image]
[image]

Hello comunity, thanks for making this greate software, I have solidworks background and I am learning FreeCAD now. I am trying to make an additive helix in part design workbench, It deforms the shape as I show in the images. These are the error messages:
 
23:29:05  0 <App> Document.cpp(4327): Failed to recompute Sin_nombre#Helix: gp_Dir() - input vector has zero norm
23:29:05  Recompute failed! Please check report view.
23:36:44  458.663 <App> Document.cpp(4327): Failed to recompute Sin_nombre1#Helix: gp_Dir() - input vector has zero norm
23:36:44  Recompute failed! Please check report view.
23:43:52  887.199 <App> Document.cpp(4327): Failed to recompute Sin_nombre1#Pocket: Null shape
23:43:52  Recompute failed! Please check report view.
23:44:01  896.398 <App> Document.cpp(4327): Failed to recompute Sin_nombre1#Pocket: Null shape
23:44:01  Recompute failed! Please check report view.
23:46:04  1018.35 <App> Document.cpp(4327): Failed to recompute Sin_nombre2#Helix: gp_Dir() - input vector has zero norm
23:46:04  Recompute failed! Please check report view.

I am using 2023.131 version in Windows 11.

realthunder, 2023-04-25:
Welcome. This most likely a CAD kernel problem. Try to make the helix radius a bit smaller so that is firmly touches the base shape.

marcx95, 2023-04-25:
thanks, It works!

closed 2023-04-25 by marcx95