realthunder/FreeCAD_assembly3 #1322: no solver backend found / no assembly found

ryanmeador, 2023-10-28:
I'm seemingly running into an installation issue with Assembly3 on my Mac.  I've installed FreeCAD 0.21.1 from Homebrew and installed Assembly3 and its dependency via the Addon Manager.  I see these lines in the report view:

```
13:19:19  1.391543 <asm3.main> init_gui.py(14): no solver backend found
13:19:55  (qt.qpa.fonts) Populating font family aliases took 309 ms. Replace uses of missing font family ".AppleSystemUIFont" with one that exists to avoid this cost. 
13:19:56  37.809071 <asm3.sys> solver.py(412): no assembly found
```

The third line is printed when I click the "Create assembly" toolbar button, and the created assembly's icon in the tree view has a 🚫 symbol on it and is dimmed out.  When I try to add constraints to my assembly, it prints the same message (and again when I try to solve said constraints; the parts never move).  I've tried reinstalling and deleting all my preferences files (per the [Wiki](https://wiki.freecad.org/Installing_on_Mac#Uninstallation)).

FreeCAD about page:
```
OS: macOS 14.0
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: (HEAD detached at 0.21.1)
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * woodworking 0.21.30486
  * Assembly3 0.12.0
```

Any ideas for how I can fix this, or info I can provide that would help diagnose it?  Thanks.

bitsmakerde, 2023-11-05:
I have the same issue :-(

bookworm840, 2024-01-07:
Same issue. Macbook pro with M2 pro chip. MacOS 14.2.1. Assembly icon is greyed out and shows 🚫, constraints don't do anything to the parts.

dannystaple, 2024-01-11:
I've found two things:
- The Mac OS freecad now uses an embedded conda Python, no the system python.
- The addon manager can suggest installing a dependancy, but it won't work, because of how/where, it's not signed and will fail to load the .so file.

I've not found a resolution, but so far it looks like the MacOS default FreeCAD download doesn't work for this.

dannystaple, 2024-01-11:
If I install this using the GUI, and accepting the python installation fo py_slvs into the Conda environment suggested in the AddON manager I get the following error if I try to import it (inside FreeCAD):
```python
>>> from py_slvs import slvs
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
  File "/Users/danielstaple/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/slvs.py", line 10, in <module>
    from . import _slvs
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.10/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
ImportError: dlopen(/Users/danielstaple/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so, 0x0002): tried: '/Users/danielstaple/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so' (code signature in <3453216D-DA76-39FF-8F7C-4548FA1B952A> '/Users/danielstaple/Library/Application Support/FreeCAD/AdditionalPythonPackages/py310/py_slvs/_slvs.so' not valid for use in process: mapped file has no cdhash, completely unsigned? Code has to be at least ad-hoc signed.)
```
[image]
[image]

ryanmeador, 2024-01-16:
Thanks for digging into this!  If I'm following your analysis correctly, this problem isn't unique to Assembly3, but it will actually affect any addon that has dependencies?  So this is a base FreeCAD issue?

dannystaple, 2024-01-16:
I think if it loads an so file, then that is likely true.

hansaplasst, 2024-04-10:
Manually downloaded [Stable 2024.04.07](https://github.com/realthunder/FreeCAD/releases/download/20240407stable/FreeCAD-Link-Stable-Linux-x86_64-py3.11-20240407.AppImage) for Linux. 

Previous threat indicated that Python module SolveSpace should be installed.
However, `pip install py_slvs` doesn't solve the issue.

`Requirement already satisfied: py_slvs in /home/user/.local/lib/python3.8/site-packages (1.0.6)`

AlbertEmil, 2024-08-20:
I can only speak for macOS, but according to this issue (https://github.com/FreeCAD/FreeCAD/issues/11071), the issue reported here seems indeed be related to singed binaries.

I had the same problems with the Assembly3 workbench (version 0.12.2) using FreeCAD 0.21.0. However, updating to FreeCAD 0.21.2, which seems to have correctly signed binaries, solved the issue for me.

collinclemons, 2024-12-31:
I am also having this issue on Windows with FreeCAD 1.0.0, after installing Assembly3 via Add-on Manager.