realthunder/FreeCAD_assembly3 #1325: Access violation when closing a document with asm3 in windows

wsteffe, 2024-01-24:
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 2024.123.0.14555 (Git shallow)
Build type: Release
Branch: HEAD
Hash: 0c0f69ad4806682096b51f7669d5f1cb83f9c7d7
Python 3.11.7, Qt 5.15.6, Coin 4.0.1, Vtk 9.2.5, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
[/code]

[image]

No error is produced in Ubuntu with the same release and same document.

The error is produced also when I am closing the document without saving (discarding changes).

realthunder, 2024-01-24:
Could you please send me the file, through email if possible?

wsteffe, 2024-01-24:
Yes I may send it. The only problem is that the linked parts are spread across several directories which contain other things. Is there a FC command which can package the assembly together with the linked subassembies and parts into a single archive ?

realthunder, 2024-01-25:
You can right click in tree view and choose `Link actions -> Import all links`.