realthunder/FreeCAD_assembly3 #1331: [Question] Crowdin translations

kaktusus, 2024-02-29:
I'd like to revisit the topic of using Crowdin for translating your workbench.
Some time ago, you were asked about this.
Now we have a new situation because I can assist you in maintaining and managing translations on Crowdin if you don't have time for it. :wink: 

So, I'd like to ask if you agree to upload the source translation file to Crowdin? 
This way, translations can be done by many nationalities.

Existing translations can also be imported.

hasecilu, 2024-03-11:
From what I have seen on the main FreeCAD Crowdin project almost every day FreeCAD is being translated, on the addons project only 7 external WB are there from more than 100!.
The point is that having the same people that translates main FreeCAD translating the addons will help end users by having a level of consistency on the whole FreeCAD environment.

I'm sure doing this fellow translators that are already helping FreeCAD ecosystem would notice the translation support more easily on Crowdin that here on GitHub.

realthunder, 2024-08-05:
> So, I'd like to ask if you agree to upload the source translation file to Crowdin?
> This way, translations can be done by many nationalities.

Sure, by all means to it, and thank you for that. Do I need to do anything on my side?

kaktusus, 2024-08-06:
Nothing in particular
you've already done it :wink: 
you have expressed approval,

luzpaz, 2024-09-10:
What's left to do here ?

hasecilu, 2024-09-10:
Translators should advance on their languages and then, if possible, test the translations. Spanish translations are still working so I don't think there is a change needed. When hopefully some languages are finished @kaktusus or someone else can bring the translations here.