realthunder/FreeCAD_assembly3 #1342: user.cfg invalid when using (old <<Sketch>>) Syntax for References to other Constraints

gneiss15, 2025-02-27:
As said in the title. Using a Constraints like:

=2 * <<Sketch>>.Constraints.Base

(Base is the Alias/Name of a Constraint in Sketch)

It works until next start of FreeCad.

When examaining what was written into user.cfg I found a line:
<FCText Name="Hist1">=2 * <<Sketch>>.Constraints.Base</FCText>

On the next start freecad seams to diside that user.cfg is invalid an creates a new one :-((

If I use the never Syntax:
=2 * Sketch.Constraints.Base)
All is good.

So I suggest that the old syntax is replaced before writing to user.cfg

realthunder, 2025-02-28:
Could you please attach a screenshot on the UI where you enter that expression?

gneiss15, 2025-03-03:
It happens on all formular entry fields..

[image]